ISPC-TST-EM2022/Grupo1 #3: Consignas para todos

FedeBirge, 2022-08-05:
Consignas particulares para todos:

- Hay 3 archivos Word en la carpeta Proyecto 2 para desarrollar los temas,
  los tres están cargados con los títulos y subtítulos a desarrollar por cada uno( se irán creando las 
issues y asignando),
  para mejor comprensión. En la revisión final los cargaremos como PDF.
- Formato a usar en el documento(ya están con margen estrecho):
	Fuente: Arial Unicode MS
		Titulo: tamaño 14 en negrita. Alineación centrada.
		Subtitulo: tamaño 12 negrita. Alineación justificada.
		Desarrollo: tamaño 11. Alineación justificada.
- Otras consideraciones:
	-Al traducirse automáticamente el contenido se generan inconsistencias,
	por favor leer y darle sentido la texto, de paso nos aprendemos el contenido.
	y colaborar con el formato please.
NOTA importante: debemos trabajar haciendo ramas(branch) para luego unir todo,
ES IMPORTANTE QUE ASIMILEMOS LOS CONOCIMIENTOS DE GIT Y GITHUB

GiseRomero, 2022-08-06:
Bueno, Fede entendido, gracias por hacer esto

Emygut, 2022-08-07:
Hola a Todos. Fede comprendido, gracias por haber organizado.

Domi74, 2022-08-07:
buen dia que se tiene que hacer estuve con finales y ahora bueno me pondre a hacer y ayudar al grupo

Emygut, 2022-08-07:
Hola, Buen Dia. Ya cargue en el documento word la Descripción de los Modos
de funcionamiento de AVR ADC, Modo diferencial AVR ADC, Modo de reducción
de ruido AVR ADC. Cualquier novedad me avisan que faltaría.

marlufer, 2022-08-07:
Hola a todos. Repasando que han hecho chicos, poniendome al dia

closed 2022-08-08 by FedeBirge